107-systems/107-Arduino-Cyphal #50: Create use case to demonstrate how to use 107-Arduino-UAVCAN to build a UAVCAN node

aentinger, 2020-11-18:
Suggestion by @pavel-kirienko in UAVCAN Weekly Dev Call 2020/11/18.

Possibly a GPS node using @generationmake [ElaraGNSSMKRShield](https://github.com/generationmake/ElaraGNSSMKRShield)? Further suggestions welcome.

pavel-kirienko, 2020-11-18:
I would look here for inspiration: http://raccoonlab.tilda.ws/uavcan. These have nothing to do with Arduino though, but they may be considered representative of the device types commonly required by new UAVCAN adopters.

[image]

Re GNSS specifically, having an RTK-capable system would be a great way to showcase the connectivity advantages offered by UAVCAN. One could seamlessly stream RTCM from an Internet-connected base station via the Internet bridge, provided that there is a modem node available in the network. Although, on the other hand, this use case is somewhat more sophisticated so I am not sure if it is an appropriate choice for the first demo.

[image]

pavel-kirienko, 2020-11-18:
Let me also tag @frontw here. Roman, I am not sure but there is a chance that you might be interested in this effort. At the UAVCAN dev call today we were discussing how we can leverage the vast Arduino ecosystem to help us promote UAVCAN **v1** with the help of @aentinger 's project (this one). This might be relevant for you as long as you are not targeting high-SIL applications.

vivienJ, 2020-11-18:
For something less sophisticated, a distance sensor? Easy to understand, used on MC, FW and VTOL and with an I2c interface as a lot of device in a UAV

aentinger, 2020-11-19:
@pavel-kirienko and @frontw : I love the graphic :rocket: Although I think component-wise it would be better to directly use UAVCAN controlled ESCs than having a UAVCAN to PWM adapter.

That being said I've suggested a GNSS node because I don't want to develop any hardware just for demoing `107-Arduino-UAVCAN` suitability to easily build UAVCAN connected devices (@vivienJ). This way the demo UAVCAN node can be assembled by stacking a couple of already available Arduino boards, e.g. `Arduino MKR Zero`, `MKR CAN Shield` and @generationmake 's [`ElaraGNSSMKRShield`](https://github.com/generationmake/ElaraGNSSMKRShield).

A ToF distance sensor might still be a an even easier demo case, as part of the [Viper Quadcopter Project](https://github.com/107-systems/107-Arduino-Viper) I'm working on an Arduino driver for the AMS [TMF8801](https://github.com/107-systems/107-Arduino-TMF8801), but @generationmake would have to build a decent breakout board for it because the one shipped by AMS is s**t (and expensive). This would require more work however, as the TMF8801 library is far from finished.

aentinger, 2020-11-24:
A first very rough template of a simple UAVCAN node exists [here](https://github.com/107-systems/UAVCAN-GNSS-Node). It does nothing but publishing a heartbeat message once per second. Be aware that the sketch will be subject to considerable change, as this is just a very rough first draft.

manmountain, 2020-12-02:
I wrote a basic node example. You can find it here: https://github.com/manmountain/107-Arduino-UAVCAN/blob/master/examples/UAVCAN-Basic_Node/UAVCAN-Basic_Node.ino in my fork of your project. I dont know if it works, I dont have the hardware to test it :) .

It should handle the basics of node responsibilities. @pavel-kirienko have I missed something mandatory? 

TODO: send some sensor value to subscribers

aentinger, 2020-12-02:
Good morning @manmountain :wave: ☕ Your basic node is looking good 👍 It would be cool if you could test it, but you need minimum `MKR Zero` + `MKR CAN Shield` + USB-CAN-Dongle (or your build your own CAN sniffer from existing hardware).

manmountain, 2020-12-02:
Gordmorning @aentinger 

I will see if I can get my hands on a mcp2515. I only have a mcp2517fd board availible right now. If anyone got the required hardware please feel free to test the example

aentinger, 2020-12-02:
Also [this construct](https://github.com/manmountain/107-Arduino-UAVCAN/blob/master/examples/UAVCAN-Basic_Node/UAVCAN-Basic_Node.ino#L125-L129) won't fly as
```C++
while(uavcan.transmitCanFrame()) { }
```
is tasking the MCP2515 or whatever else provides the CAN PHY layer with transmission, however transmission is not guaranteed at this port. Since you do a reset afterwards the MCP2515 will re-initialize (also reset during that process) and your CAN frames in the process of tranmission will be dropped.

manmountain, 2020-12-02:
Thanks, i will look into it.

pavel-kirienko, 2020-12-02:
> It should handle the basics of node responsibilities. @pavel-kirienko have I missed something mandatory?

@manmountain you didn't, but there is one recommended detail which is the register access service (we also call it "the Register API", informally). It defines important standard registers intended to configure the subject-ID of the messages you publish or subscribe to:

https://github.com/UAVCAN/public_regulated_data_types/blob/d874225ea4b5a79a4639bcfcb9096ccd943138a9/uavcan/register/384.Access.1.0.uavcan#L87-L125

In a reusable example, it is quite important to support it.

aentinger, 2021-01-11:
I'll be closing this for now as two reference implementations ([UAVCAN-ToF-Distance-Sensor-Node](https://github.com/107-systems/UAVCAN-ToF-Distance-Sensor-Node) and [UAVCAN-GNSS-Node](https://github.com/107-systems/UAVCAN-GNSS-Node)) are already work-in-progress.

closed 2021-01-11 by aentinger