107-systems/107-Arduino-Cyphal #68: 'exception_ptr' is not a member of 'std'

kyle-pawlowski, 2021-01-18:
I'm using this library for an ARM chip with the SKU ATSAMC21E18A. Whenever I try to compile, I get the error, _'exception_ptr' is not a member of 'std'_ from the _catch.hpp_ file. I feel like the issue probably has something to do with the compiler I'm using. I am using the latest toolchain for Atmel Studio. I'm using the GNU compiler with the _-std=c++0x_ option. I also tried using _-std=c++11_ and _std=c++14_. The Catch2 github says it works with C++11 and exception_ptr is a C++11 feature. What is the compiler that you used when compiling this library? I'm not using an Arduino, but from looking at the code, I think this library is hardware agnostic. Thanks for your help.

aentinger, 2021-01-19:
Good morning :wave: :coffee: Please be advised that this is an Arduino library which is only supported in context with using the Arduino toolchain. Anything else and you are on your own. That being said I believe you experience this issue because your toolchain is picking up the test code within the `extras/test` subfolder. Just delete it and you should be good.

closed 2021-01-19 by aentinger

kyle-pawlowski, 2021-01-19:
Thanks! I appreciate the response!