107-systems/107-Arduino-Cyphal #69: not compiling with esp32

generationmake, 2021-01-24:
The examples of the library are not compiling with esp32.

It gives the following error:

```
In file included from C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN\src/ArduinoUAVCAN.h:28:0,

                 from C:\Users\Bernhard\AppData\Local\Temp\arduino_modified_sketch_951541\UAVCAN-Heartbeat-Publish.ino:15:

C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN\src/utility/convert.hpp:21:18: error: expected '{' before '::' token

 namespace arduino::_107_::uavcan

                  ^

C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN\src/utility/convert.hpp:21:20: error: '_107_' in namespace '::' does not name a type

 namespace arduino::_107_::uavcan

                    ^

UAVCAN-Heartbeat-Publish:127:1: error: expected '}' at end of input

 }

 ^

exit status 1

expected '}' at end of input
```

To solve this error is somehow beyond my possibilities...

aentinger, 2021-01-24:
Looks to me like nested namespaces are not (yet) supported on the ESP32 platform. @per1234 may I ask you to add CI for ESP32 in a separate PR?

generationmake, 2021-01-24:
I did further testing. The above error message was using the latest release installed via Arduino Library Manager.

Now I updated to the master branch of github. This gives a slightly different error message but the program still does not compile:

```
In file included from C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN\src/types/uavcan/node/ExecuteCommand.1.0.hpp:19:0,

                 from C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN\src/ArduinoUAVCANTypes.h:15,

                 from C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN\src/ArduinoUAVCAN.h:24,

                 from C:\Users\Bernhard\AppData\Local\Temp\arduino_modified_sketch_855220\UAVCAN-Heartbeat-Publish.ino:15:

c:\users\bernhard\documents\arduino\libraries\107-arduino-uavcan\src\utility\convert.hpp:21:18: error: expected '{' before '::' token

 namespace arduino::_107_::uavcan

                  ^

c:\users\bernhard\documents\arduino\libraries\107-arduino-uavcan\src\utility\convert.hpp:21:20: error: '_107_' in namespace '::' does not name a type

 namespace arduino::_107_::uavcan

                    ^

In file included from C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN\src/ArduinoUAVCANTypes.h:15:0,

                 from C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN\src/ArduinoUAVCAN.h:24,

                 from C:\Users\Bernhard\AppData\Local\Temp\arduino_modified_sketch_855220\UAVCAN-Heartbeat-Publish.ino:15:

C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN\src/types/uavcan/node/ExecuteCommand.1.0.hpp: In member function 'void arduino::ExecuteCommand_1_0::Response::operator=(arduino::ExecuteCommand_1_0::Response::Status)':

C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN\src/types/uavcan/node/ExecuteCommand.1.0.hpp:116:28: error: 'arduino::_107_' has not been declared

     data.status = arduino::_107_::uavcan::to_integer(status);

                            ^

In file included from C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN\src/ArduinoUAVCANTypes.h:16:0,

                 from C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN\src/ArduinoUAVCAN.h:24,

                 from C:\Users\Bernhard\AppData\Local\Temp\arduino_modified_sketch_855220\UAVCAN-Heartbeat-Publish.ino:15:

C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN\src/types/uavcan/node/Heartbeat.1.0.hpp: In member function 'void arduino::Heartbeat_1_0::operator=(arduino::Heartbeat_1_0::Health)':

C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN\src/types/uavcan/node/Heartbeat.1.0.hpp:78:34: error: 'arduino::_107_' has not been declared

     data.health.value = arduino::_107_::uavcan::to_integer(health);

                                  ^

C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN\src/types/uavcan/node/Heartbeat.1.0.hpp: In member function 'void arduino::Heartbeat_1_0::operator=(arduino::Heartbeat_1_0::Mode)':

C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN\src/types/uavcan/node/Heartbeat.1.0.hpp:83:32: error: 'arduino::_107_' has not been declared

     data.mode.value = arduino::_107_::uavcan::to_integer(mode);

                                ^

C:\Users\Bernhard\AppData\Local\Temp\arduino_modified_sketch_855220\UAVCAN-Heartbeat-Publish.ino: At global scope:

UAVCAN-Heartbeat-Publish:127:1: error: expected '}' at end of input

 }

 ^

Bibliothek SPI in Version 1.0 im Ordner: C:\Users\Bernhard\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI  wird verwendet

Bibliothek 107-Arduino-UAVCAN in Version 1.1.0 im Ordner: C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-UAVCAN  wird verwendet

Bibliothek 107-Arduino-MCP2515 in Version 1.2.2 im Ordner: C:\Users\Bernhard\Documents\Arduino\libraries\107-Arduino-MCP2515  wird verwendet

exit status 1

expected '}' at end of input

```

aentinger, 2021-01-25:
It's still the same error afaik - `namespace arduino::_107_::uavcan` is a so called nested namespace which is only supported from C++17 forward.

PR #71 (2021-01-25): Fixing build issues when using this library with ESP32.
ESP32 Core does not support C++14, ergo no nested namespaces are supported. This commit changes the code to conventional namespaces therefore allowing CI to pass for ESP32.

This fixes #69.
closed 2021-01-25 by aentinger