107-systems/107-Arduino-Cyphal #73: Required performace..

AndKe, 2021-01-29:
I plan to make a BMS using Atmega32u4 (16Mhz) and SN65HVD230 fro ArduPolit.
would you can the Atmage32u4 is enough, or should I go straight for STM32F103 (or some other, small-footprint STM?)

echoGee, 2021-01-29:
atmega32u4 is a not the best for CAN. low on memory etc. Our earlier versions where on this chip, but newest v4 is the Atsam series 

https://rotoye.com/batmon/BatMon-v4-02-with-SMBUS-CAN-FD-LIN-2-packs-p273850043. The entire firmware is available with the purchase of the board.

generationmake, 2021-01-29:
I don't know which CAN controller you are going to use. If you plan to use MCP2515 with the [107-Arduino-MCP2515](https://github.com/107-systems/107-Arduino-MCP2515) library please note that this only works with samd,mbed,esp32.
So I suggest to stay with these architectures.
At the moment the 107-Arduino-UAVCAN library is also only tested with samd and mbed.

aentinger, 2021-01-30:
Hi @AndKe :wave: Both @echoGee and @generationmake are correct, `atmega32u4` might be problematic due too low RAM and is definitely problematic because `ArduinoCore-AVR` does not support the C++ standard used within this library.

closed 2021-01-30 by aentinger