107-systems/107-Arduino-Cyphal #74: generates corrupted CAN messages

generationmake, 2021-01-30:
The library generates corrupted CAN messages when received with a MCP2518FD:

`ip -details -statistics link show type can`

shows 

```
4: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state ERROR-PASSIVE (berr-counter tx 0 rx 131) restart-ms 0 
	  bitrate 250000 sample-point 0.875 
	  tq 50 prop-seg 34 phase-seg1 35 phase-seg2 10 sjw 1
	  mcp251xfd: tseg1 2..256 tseg2 1..128 sjw 1..128 brp 1..256 brp-inc 1
	  mcp251xfd: dtseg1 1..32 dtseg2 1..16 dsjw 1..16 dbrp 1..256 dbrp-inc 1
	  clock 20000000
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          0          0          2061       9048       0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 
    RX: bytes  packets  errors  dropped overrun mcast   
    3400308    639469   0       18      0       0       
    TX: bytes  packets  errors  dropped carrier collsns 
    0          0        0       0       0       0       
```

Testsetup:
* Arduino MKR WiFi with Arduino CAN MKR Shield, Sketch: UAVCAN-Heartbeat-Publish example without modifications
* Raspberry Pi with MCP2518FD on SPI bus

Findings so far:
* the sketch generates the receive errors as shown above.
* using [107-Arduino-MCP2515](https://github.com/107-systems/107-Arduino-MCP2515) and the example from teh Readme alone doesn't generate any errors
* using an MCP2515 on the Raspberry Pi side doesn't generate any errors
* using other CAN libraries like https://github.com/sandeepmistry/arduino-CAN doesn't generate any receive errors

Sorry that I can't give a better error description.I have tested several things but nothing worked.
Maybe someone with a CAN analyzer could help here.

generationmake, 2021-01-30:
same on esp32 with MCP2515

aentinger, 2021-02-01:
Good morning @generationmake :wave: ☕ 
Thank you very much for reporting your findings. I've already cross-checked the baud rate register configurations with other libraries and find no difference. Which version of [107-Arduino-MCP2515](https://github.com/107-systems/107-Arduino-MCP2515) have you been using for your tests? There was a recently fixed bug which could (maybe) explain the issue you are seeing here: https://github.com/107-systems/107-Arduino-MCP2515/commit/5c33302c163ec4718b17fe6982bc5dbccfe8eff0. If you haven't already, can you retry with the latest version as stored on [107-Arduino-MCP2515:master:HEAD](https://github.com/107-systems/107-Arduino-MCP2515/commits/master)?

generationmake, 2021-02-01:
I think I already tested with one of the latest version but I will check again with THE latest version.

generationmake, 2021-02-04:
So, here are my results.
I took the latest code from the master branch on github of 107-Arduino-MCP2515 and 107-Arduino-UAVCAN.
Turns out, the error comes with both implementations.

### 107-Arduino-MCP2515

#### code

```
/*
 * This example shows periodic transmission of a UAVCAN heartbeat message via CAN.
 *
 * Hardware:
 *   - Arduino MKR family board, e.g. MKR VIDOR 4000
 *   - Arduino MKR CAN shield
 */

/**************************************************************************************
 * INCLUDE
 **************************************************************************************/

#include <SPI.h>

#include <ArduinoUAVCAN.h>
#include <ArduinoMCP2515.h>

/**************************************************************************************
 * CONSTANTS
 **************************************************************************************/

static int const MKRCAN_MCP2515_CS_PIN  = 3;
static int const MKRCAN_MCP2515_INT_PIN = 7;

/**************************************************************************************
 * FUNCTION DECLARATION
 **************************************************************************************/

void    spi_select      ();
void    spi_deselect    ();
uint8_t spi_transfer    (uint8_t const);
void    onExternalEvent ();
bool    transmitCanFrame(CanardFrame const &);

/**************************************************************************************
 * GLOBAL VARIABLES
 **************************************************************************************/

ArduinoMCP2515 mcp2515(spi_select,
                       spi_deselect,
                       spi_transfer,
                       micros,
                       nullptr,
                       nullptr);

ArduinoUAVCAN uavcan(13, transmitCanFrame);

Heartbeat_1_0 hb;

/**************************************************************************************
 * SETUP/LOOP
 **************************************************************************************/

void setup()
{
  Serial.begin(9600);
//  while(!Serial) { }

  /* Setup SPI access */
  SPI.begin();
  pinMode(MKRCAN_MCP2515_CS_PIN, OUTPUT);
[... 196 lines cut ...]
 (1612422240.919898)  can0  RX - -  107D550D   [8]  68 00 00 00 00 00 00 EA   'h.......'
 (1612422241.920883)  can0  RX - -  107D550D   [8]  69 00 00 00 00 00 00 EB   'i.......'
 (1612422242.921931)  can0  RX - -  107D550D   [8]  6A 00 00 00 00 00 00 EC   'j.......'
 (1612422243.922907)  can0  RX - -  107D550D   [8]  6B 00 00 00 00 00 00 ED   'k.......'
 (1612422244.930879)  can0  RX - -  107D550D   [8]  6C 00 00 00 00 00 00 EE   'l.......'
```

#### statistics

```
4: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state ERROR-WARNING (berr-counter tx 0 rx 125) restart-ms 0 
	  bitrate 250000 sample-point 0.875 
	  tq 50 prop-seg 34 phase-seg1 35 phase-seg2 10 sjw 1
	  mcp251xfd: tseg1 2..256 tseg2 1..128 sjw 1..128 brp 1..256 brp-inc 1
	  mcp251xfd: dtseg1 1..32 dtseg2 1..16 dsjw 1..16 dbrp 1..256 dbrp-inc 1
	  clock 20000000
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          0          0          2          10         0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 
    RX: bytes  packets  errors  dropped overrun mcast   
    1024       128      0       32      0       0       
    TX: bytes  packets  errors  dropped carrier collsns 
    0          0        0       0       0       0       
```

generationmake, 2021-02-04:
OK, the error also exist with the MKR CAN Shield on an ESP32.
Maybe my CAN shield  is broken?

generationmake, 2021-02-05:
This seems to be related to the receiver settings of the MCP2518FD. Setting the sample point from 0.875 to 0.75 seems to solve this issue.
Maybe it is also related to the CAN transceiver on the MKR CAN Shield.

aentinger, 2021-02-05:
Great finding. This would mean that this issue isn't actually an issue at all. I'm going to close it, please reopen if you disagree.

closed 2021-02-05 by aentinger