107-systems/107-Arduino-Cyphal #83: Fix compilation for ESP32

aentinger, 2021-02-09:
Blocked by https://github.com/espressif/arduino-esp32/issues/4771.

chacalnoir, 2021-04-09:
I ran into this issue as well. I was able to work around it by adding:

using namespace std;

before:

#include <ArduinoUAVCAN.h>

in my source file.  That allows me to compile for an Adafruit Huzzah32.  Note that I do not consider this a "fix" because (1) adding using namespace std in a header file (where I need to include ArduinoUAVCAN.h) is a bad idea, in general, and (2) I should not need to add a using directive outside the header file that actually needs the using directive (i.e the file that uses isfinite should be the file that includes the using directive, if it is required). However, it seems to be a possible workaround, given that my other attempts:
(1) changing all isfinite calls to std::isfinite resulted in a macro expansion that caused a syntax and compilation error or
(2) removing the <math.h> include or removing the isfinite macro definition resulted in a compilation error because isfinite was not found in the std namespace.

aentinger, 2021-04-12:
Thank you for reporting your findings @chacalnoir :wave: This made is possible to design a workaround for me to make that library compile for ESP32 again (#102) without requiring user sketch modifiction.

PR #102 (2021-04-12): Workaround for erroneous macro expansion (isfinite) on ESP32
Undefining isfinite and replacing its usage with std::finite fixes macro expansion issues when compiling for ESP32.

This fixes #83.
closed 2021-04-12 by aentinger