107-systems/107-Arduino-Cyphal #94: Update wiki to instruct users to edit the nunavut generated header files due to include path

echoGee, 2021-02-12:
https://github.com/107-systems/107-Arduino-UAVCAN/wiki has intructions to make header files. But the instructions to edit the header files need to be added for completeness and accuracy

pavel-kirienko, 2021-02-12:
It would also be nice to link http://nunaweb.uavcan.org

aentinger, 2021-02-13:
I think you can both see that the wiki is currently more a personal reminder than for general consumption :wink: Would one of you be able to supply a first draft?

echoGee, 2021-02-13:
Draft of ?

aentinger, 2021-02-15:
I meant a draft of desirable documentation. I'm going to be honest, I'm currently drowning in work. Consequently the wiki documentation is quite low on priority. Furthermore, this specific bit of information (that is how to add new wrappers) is quite important so it might end up being moved to the README instead. If one of you does have the time or the drive to write that up then please be my guest, as I fear I won't get to it anytime soon.

PR #96 (2021-02-16): Document the process for adding currently missing DSDL types.
This fixes #94.
commit cdb03082 (2021-02-16): Document the process for adding currently missing DSDL types. (#96)

* Document the process for adding currently missing DSDL types.

This fixes #94.

* Streamline installation instructions for nunavut.

* Be consistent and always compile for any target endianness.
closed 2021-02-16 by aentinger