107-systems/107-Arduino-Cyphal #99: Fix: Assign fixed port IDs for DSDL wrappers which have a fixed port ID assigned

aentinger, 2021-02-24:
All existing DSDL wrappers stored in [src/wrappers ](https://github.com/107-systems/107-Arduino-UAVCAN/tree/master/src/wrappers) need to be checked against the DSDL specification in [public_regulated_data_types](https://github.com/UAVCAN/public_regulated_data_types). Any type that has a fixed-port ID (can be identified by a number prefixed to the type name), e.g. [409.Write.1.0.uavcan](https://github.com/UAVCAN/public_regulated_data_types/blob/master/uavcan/file/409.Write.1.0.uavcan), the corresponding wrapper need to be modified as shown below:
```diff
-template <CanardPortID ID>
class Request
{

public:

  uavcan_file_Write_Request_1_0 data;

-  static constexpr CanardPortID       PORT_ID          = ID;
+  static constexpr CanardPortID       PORT_ID          = uavcan_file_Write_1_0_FIXED_PORT_ID_;

...
-template <CanardPortID ID>
class Response
{

public:

  uavcan_file_Write_Response_1_0 data;

  static constexpr CanardPortID       PORT_ID          = ID;
+ static constexpr CanardPortID       PORT_ID          = uavcan_file_Write_1_0_FIXED_PORT_ID_;
```

pavel-kirienko, 2021-02-24:
One might want to use a type with a fixed port-ID over a non-fixed port. Why don't you just say `template <CanardPortID ID = uavcan_file_Write_1_0_FIXED_PORT_ID_>`?

aentinger, 2021-02-24:
This would work :+1: My only gripe is that the whole template concept is not future-proof when it comes to changing ports via register API during runtime. You've got a smart way out of that? (I wanted to address it only after fixing this).

pavel-kirienko, 2021-02-24:
> You've got a smart way out of that?

Ah yes. Irrespective of my above comment, templates shouldn't be used here at all because port-IDs are not meant to be hard-coded.

aentinger, 2021-02-24:
I'll update above comment with your suggestions and let's fix the hard-coded port-id another time ;)

aentinger, 2021-07-02:
Fixed by #109.

closed 2021-07-02 by aentinger