107-systems/107-Arduino-Cyphal #100: Add UAVCAN logo to README

aentinger, 2021-03-01:
@pavel-kirienko would that be okay?

pavel-kirienko, 2021-03-01:
Of course. We only regulate usage of the UAVCAN Conformity Mark --- using that without permission would be illegal, but the regular UAVCAN logo can be used freely. We also have sick ASCII art:

```
 __   __   _______   __   __   _______   _______   __   __
|  | |  | /   _   ` |  | |  | /   ____| /   _   ` |  ` |  |
|  | |  | |  |_|  | |  | |  | |  |      |  |_|  | |   `|  |
|  |_|  | |   _   | `  `_/  / |  |____  |   _   | |  |`   |
`_______/ |__| |__|  `_____/  `_______| |__| |__| |__| `__|
    |      |            |         |      |         |
----o------o------------o---------o------o---------o-------
```

aentinger, 2021-03-02:
That's pretty neat :wink: I think I've found the UAVCAN logo in [eps](https://github.com/UAVCAN/specification/blob/master/specification/uavcan_logo.eps) ... can you point me towards a JPG/PNG version?

pavel-kirienko, 2021-03-02:
We only have SVG here: https://github.com/UAVCAN/uavcan.org/blob/master/app/static/images/logo.svg

PR #101 (2021-03-02): Adding UAVCAN logo on the README top/right corner.
This fixes #100.
closed 2021-03-02 by aentinger