107-systems/107-Arduino-Cyphal #120: Does the configurable runtime IDs is supported? If so is there any example code?

pepeRossRobotics, 2021-08-06:
Hello
I am familiarising myself with this project, I am fairly new to CAN so still getting my feet wet. Sorry if some (all) of my terms are incorrect. 

I would like to know if this library allows for the IDs to be set during runtime (to have multiple nodes with the same code) by other devices in the CAN bus.
 
If so, how does this works? Is there any example on how to do it? Does it happens automatically?

Many thanks.
-Pepe

aentinger, 2021-08-10:
Hi :wave: I'm currently on vacation. Even when not in vacation support requests are only answered once / week at my personal discretion.

pavel-kirienko, 2021-08-10:
I am not a developer of this library but let me answer for Alexander *(who is diligently reading his mail despite being on vacation --- we can all take a page from his book)*.

>I would like to know if this library allows for the IDs to be set during runtime (to have multiple nodes with the same code) by other devices in the CAN bus.

You can set node-IDs at runtime but not port-IDs. This is a [known limitation](https://github.com/107-systems/107-Arduino-UAVCAN/issues/99#issuecomment-785237425) that we discussed earlier but it was never publicly documented, which was an omission on our part (as I said, I am not a developer, but I consider myself to be at least partially responsible for any UAVCAN-related shortcomings out there in the world).

If you feel like coding, then perhaps Alexander would appreciate some help in getting this deficiency corrected. Or maybe he wouldn't, who knows. At any rate, you could always just use libcanard directly instead of this library --- its raw C API is lacking in the sense of ergonomics (because it's C, what did you expect) but it does not limit you to fixed port-IDs.

If you have any general questions about UAVCAN then the [UAVCAN Forum](https://forum.uavcan.org/) might be a better place for asking them!

pepeRossRobotics, 2021-08-11:
Many thanks for the reply. I will check what makes more sense for my project, ge my feet more wet into this UAVCAN world and see whats my next move. :)

closed 2021-08-11 by pepeRossRobotics

aentinger, 2021-08-13:
Thank you @pavel-kirienko :wave: I could not have it said better 😍 

@pepeRossRobotics Open-Source does not mean that the author of any piece of code is your personal support servant. Some lifting (i.e. writing clear and concise questions) from your side can and is expected.