107-systems/107-Arduino-Cyphal #121: Compiling for rp2040 using earlephilhower/arduino-pico by removing LockGuards

pepeRossRobotics, 2021-08-12:
Hello I was having issues getting my Pi-pico running properly on the ArduinoCre-mbed for the pi-pico as the serial interface seems to have some issues ad my code was hanging (i think it was crashing the chip)
So I decided to try to compile this library for the pi-pico using the `earlephilhower/arduino-pico` project.

I was not able to do it straight out of the box, but wen I removed the calls for the `LockGuard` (that seems to be mbed only) I was able to compile it and even receive frames.

I am yet to test more functionality as I am still learning and setting up.

I would like to know if I will be getting into some kind of trouble by removing the `LockGuard` calls even if I am not using the mbed library. I assume that the RTOS of ArdunoCore make `LockGuard` needed but I might be just 100% wrong.

If this approach is not wrong, could this be something that the library incorporate? Checking if Arduino is running mbed, and if not, just removing the Lockuards?

aentinger, 2021-08-13:
Hi @pepeRossRobotics :wave:

I'm still on vacation 🍹 A couple of pointers though:

1) This library does not support [earlephilhower/arduino-pico](https://github.com/earlephilhower/arduino-pico). Consequently the `arduino-pico` core is not listed in either [README](https://github.com/107-systems/107-Arduino-UAVCAN/blob/master/README.md) or within [library.properties](https://github.com/107-systems/107-Arduino-UAVCAN/blob/master/library.properties#L9).

2) If you disable locking AND you don't know what you are doing (i.e. accessing the API of this library from both ISR context and normal execution - which is quite a common scenarion) you will suffer from race conditions. Proceed on your own risk.

3) If you want this library to support [earlephilhower/arduino-pico](https://github.com/earlephilhower/arduino-pico) than I suggest to provide a `CritSec-pico.c/cpp` and create a pull request. I can help with sorting out the details.

pepeRossRobotics, 2021-08-20:
Hello, 

I made a small `CritSec-rpipico.cpp` file that uses the built in `noInterrupts()` and `interrupts()` built in function.
Do you think this can work, I am testing on my small system while still getting familiarised with UAVCAN so I might be oversimplifying things.
```
#include "CritSec.h"
#ifdef ARDUINO_RASPBERRY_PI_PICO
#include "Arduino.h"

/**************************************************************************************
 * FUNCTION DEFINITION
 **************************************************************************************/

extern "C" void crit_sec_enter()
{
  noInterrupts();
}

extern "C" void crit_sec_leave()
{
  interrupts();
}

#endif /* ARDUINO_RASPBERRY_PI_PICO */
```
I can make a PR if you want, but I think that requires some permissions for me to push into the repository.

aentinger, 2021-08-20:
Please create a PR :+1: (And no, you don't need some permissions). Details see [here](https://docs.github.com/en/github/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request).

pepeRossRobotics, 2021-08-20:
PR done -> https://github.com/107-systems/107-Arduino-UAVCAN/pull/125

closed 2021-08-20 by pepeRossRobotics