107-systems/107-Arduino-Cyphal #131: Heartbeat in ToF-Sensor Example is transmitted more frequently than 1 Hz

joelsa, 2021-09-12:
The prev_heartbeat is set to 0 at each iteration of the loop:

https://github.com/107-systems/107-Arduino-UAVCAN/blob/5e03e6e795a65d0b3507560f82381d3f70a165e8/examples/UAVCAN-ToF-Distance-Sensor-Node/UAVCAN-ToF-Distance-Sensor-Node.ino#L171

Also, do you plan on completing the ToF-Sensor example in order to publish ranges instead of just the heartbeat? Please let me know if I can contribute something there.

joelsa, 2021-09-12:
> Also, do you plan on completing the ToF-Sensor example in order to publish ranges instead of just the heartbeat? Please let me know if I can contribute something there.

I guess this is not possible, as there is no range sensor message in https://github.com/UAVCAN/public_regulated_data_types?

pavel-kirienko, 2021-09-12:
The message is there: https://github.com/UAVCAN/public_regulated_data_types/blob/b02e6899a319ddefbab41b820d167c95dd00174d/uavcan/si/sample/length/Scalar.1.0.uavcan

aentinger, 2021-09-13:
>Also, do you plan on completing the ToF-Sensor example in order to publish ranges instead of just the heartbeat?

Good Morning @joelsa :wave: ☕ 

I fully intend to have the ToF-Example publish both heartbeat and distance information. Alas, there are a couple of todo-items on my list before I get to this. It would be very helpful if you could take care https://github.com/107-systems/107-Arduino-UAVCAN/issues/98 as this is a blocker on transmitting the range information.

PS: You should have been forced to select either submit a bug report or a feature request upon creating this issue, each coming with it's distinct template. Why am I not seeing either template in your issue? These templates exist to ease the work of unpaid open-source developers so I kindly ask you to abide by them in the future.

joelsa, 2021-09-13:
> PS: You should have been forced to select either submit a bug report or a feature request upon creating this issue, each coming with it's distinct template. Why am I not seeing either template in your issue?

I opened the issue not via the "issues" tab but by clicking on the line in code and hitting "reference in new issue". I did not get any option to use a template there. I think this is a GitHub usability issue and fully agree that templates should be used in any case.

aentinger, 2021-09-13:
Interesting :wink: Thank you for the clarification, from the existing documentation I was under the impression that [this line](https://github.com/107-systems/.github/blob/main/.github/ISSUE_TEMPLATE/config.yml#L1) should enforce usage of issue templates everywhere. Would be probably worth reporting to GitHub.

joelsa, 2021-09-21:
> I was under the impression that [this line](https://github.com/107-systems/.github/blob/main/.github/ISSUE_TEMPLATE/config.yml#L1) should enforce usage of issue templates everywhere. Would be probably worth reporting to GitHub.

Somebody already took the time to do so: https://github.community/t/how-to-use-issue-templates-with-reference-in-new-issue/2359

diodenschein, 2022-03-24:
I'm struggling to work out how to transmit actual pressure data not just heartbeat with this library. I think one example would help me a lot.  What's still missing for transmitting the range? Can I do anything?

PR #150 (2022-06-17): Complete ToF Distance Node example.
This fixes #131.
closed 2022-06-17 by aentinger