107-systems/107-Arduino-Cyphal #139: Correct behaviour of transmit callback?

NerdyProjects, 2021-11-05:
The transmit callback method is of boolean type and is able to push the CAN frame back into libcanard. I put together some available snippets which also included the TX feedback from the CAN controller into this:

tx callback gets called, writes packet to CAN hardware, waits until CAN hardware signals the packet has been sent and then returns with true; returns with false if there is an error/timeout with the hardware.

While we are waiting for the CAN hardware to transmit the packet, in an unlikely scenario there is 100 higher priority CAN packets receiving which would overflow my CAN hardwares RX fifo as the lock inside Arduino UAVCAN does not allow the `onCanFrameReceived` method to execute before the TX packet has been acknowledged.

So is it wrong to wait until the CAN packet is actually send? In which cases should the tx callback return false then?

pavel-kirienko, 2021-11-05:
>So is it wrong to wait until the CAN packet is actually send?

In most cases, yes, it is wrong, because the transmission latency is influenced by external factors outside of your control, such as transmission patterns of other nodes sharing the same bus with you. This does not necessarily hold for TTCAN though, but I understand this is not your case.

closed 2021-11-05 by NerdyProjects

aentinger, 2021-11-06:
It depends. The transmit callback API was written with the behaviour of the 107-Arduino-MCP2515 transmit function in mind. The MCP2515 has 3 transmit buffers, so the 107-Arduino-UAVCAN library would fill up all available transmit buffers with the MCP2515 then independently doing the actual transmission. So it depends ... on how you design the underlying functionality of your transmit callback.

Antworten: ***@***.***
Cc: ***@***.***

Closed #139<https://github.com/107-systems/107-Arduino-UAVCAN/issues/139>.