107-systems/107-Arduino-Cyphal #140: 'uavcan::node::Heartbeat_1_0' is not a template

SMEA98, 2021-11-08:
### :bug: Bug Report

<p align="center">
  [image]
</p>

**Describe the bug**

A clear and concise description of what the bug is.

**Minimum example code to replicate the bug**
```C++
/*'uavcan::node::Heartbeat_1_0' is not a template is the error I am getting. I'm not able to compile your example GNSS sketch, I am using the master repo.  Thanks!*/
```

**Minimum configuration to replicate the bug**
Please provide the configuration data used to build Snowfox:
* Build Toolchain: (e.g. Arduino IDE 1.8.13, arduino-cli 0.13.0)
* Arduino Core: (e.g. ArduinoCore-samd 1.8.9)
* Library Version (of all involved libraries): (e.g. 107-Arduino-UAVCAN 1.1.0, 107-Arduino-MCP2515 1.2.0)

aentinger, 2021-11-09:
Good morning @SMEA98 :wave: :coffee: 

You are definitely not using the code within the master branch but the latest released version, the same problem was reported [here](https://forum.uavcan.org/t/arduino-uavcan-gnss-node-example-compilation-error/1432). Also the head of master compiles just fine, I just tried ...

```bash
arduino-cli compile -v --warnings all -b arduino:samd:mkrzero examples/UAVCAN-GNSS-Node
```

However, a new release is long overdue so I shall proceed with that too so that the latest version can be installed via library manager.

aentinger, 2021-11-09:
Done :+1: [1.4.0](https://github.com/107-systems/107-Arduino-UAVCAN/releases/tag/1.4.0) has just been released and shall be available via the library manager shortly.

closed 2021-11-09 by aentinger