107-systems/107-Arduino-Cyphal #141: [Feat] Support for Arduino-STM32 based systems

vicgwee, 2021-11-11:
### ⚡ Feature Request

<p align="center">
  [image]
</p>

**Please Describe The Problem To Be Solved**
There isn't much UAVCAN support for [STM32duino](https://github.com/stm32duino/Arduino_Core_STM32)-based systems currently. Are there any plans to include compatibility for it?

aentinger, 2021-11-11:
No, there are no plans to add support for STM32duino-based systems. However, you can help it further along by starting a PR adding continuous integration for this platform. This should be quite easy by following the pattern established e.g. by the [RP2040](https://github.com/107-systems/107-Arduino-UAVCAN/blob/master/.github/workflows/compile-examples.yml#L83-L86).

aentinger, 2021-11-12:
Hi @vicgwee ... this is a whole world of hurt due to very interesting ways stm32duino treats lambda's. It already starts [here](https://github.com/107-systems/107-Arduino-MCP2515/runs/4187353139?check_suite_focus=true#step:4:55). If you want this to happen, you'll have to contribute some work yourself, i.e. fixing for https://github.com/107-systems/107-Arduino-UAVCAN/pull/143 a start.

WaldoPepper, 2022-01-02:
Hi, I would like to see this in 107...UAVCAN, too. I just do not grasp the reference to the RP2040 above, though. The RP does not have an integrated CAN controller which most off the STM's have (bxcan "macro cell"). Could this be an asset in this task, or is it more of a liability? One of the original UAVCAN repos makes reference to a STM32 CAN driver, btw.

aentinger, 2022-01-05:
Hi @WaldoPepper ... supporting STM32 is not a priority for me, however I'm willing to accept contributions. It does not matter, that the STM32 has internal CAN and the RP2040 has not, this library just needs to be provided with a [function to transmit CAN frames](https://github.com/107-systems/107-Arduino-UAVCAN/blob/master/src/ArduinoUAVCAN.h#L48) and one function needs to be called [upon the reception of CAN frames](https://github.com/107-systems/107-Arduino-UAVCAN/blob/master/src/ArduinoUAVCAN.h#L54).

If those functions are served via an external CAN module such as the MCP2515 or by a MCU internal CAN module with a suitable driver API does not matter :shrug: .

aentinger, 2023-01-02:
I'm closing this PR, supporting `stm32duino` is not a priority at the moment.