107-systems/107-Arduino-Cyphal #154: Unable To Run Service Example On Raspberry Pi Pico

AS12Dev, 2022-07-31:
### Unable To Run Service Example On Raspberry Pi Pico

Hi, I am able to build and run the heartbeat examples successfully, and have produced nodes that send and receive multiple types of messages. However, although I am able to compile the service examples, nothing is displayed on the serial monitor for either Pi Pico. I haven't altered the code aside from setting the SPI pins. Thanks in advance for your help.

**Code:**
```C++
// ------------- SPI Communication Pins -------------

constexpr unsigned int SPI_INT                  =  6;
constexpr unsigned int SPI_TX                   = 19;
constexpr unsigned int SPI_SCK                  = 18;
constexpr unsigned int SPI_CS                   = 17;
constexpr unsigned int SPI_RX                   = 16;

// --------------------------------------------------

...

SPI.setRX(SPI_RX);
SPI.setTX(SPI_TX);
SPI.setSCK(SPI_SCK);
SPI.setCS(SPI_CS);
SPI.begin();

...

pinMode(SPI_INT, INPUT_PULLUP);
attachInterrupt(digitalPinToInterrupt(SPI_INT), []() { mcp2515.onExternalEventHandler(); }, FALLING);
```

**Configuration Data Used:**
* Build Toolchain: Arduino IDE 1.8.19
* Arduino Core: [earlephilhower](https://github.com/earlephilhower) / [arduino-pico](https://github.com/earlephilhower/arduino-pico)
* Library Version: Latest release of 107-Arduino-Cyphal & 107-Arduino-MCP2515

aentinger, 2022-08-01:
Hi @AS12Dev ☕ :wave:

Thank you for your report. Honestly, this sounds very much like a physical layer problem to me. Can you narrow it down by writing a very simplified sketch just sending one singular CAN frame from one board to the next?

joerossrobotics, 2022-08-01:
Thank you for getting back to me so quickly. Sending a singular CAN frame from one pico to the other works as expected.

AS12Dev, 2022-08-01:
Thank you for getting back to me so quickly. Sending one singular CAN frame from one pico to the other works as expected.

aentinger, 2022-08-02:
Another thought: The OpenCyphal-Service-Client [fires](https://github.com/107-systems/107-Arduino-Cyphal/blob/main/examples/OpenCyphal-Service-Client/OpenCyphal-Service-Client.ino#L84) off the request only once (and immediately after startup). Could it be that the server is not up be the time the client sends the request?

closed 2022-09-14 by aentinger