107-systems/107-Arduino-Cyphal #174: [Feature Request] Remodel API to emulate ROS publish/subscribe APIs

aentinger, 2022-12-13:
[image]

### ⚡ Feature Request

Doing so would simplify the internal logic of `Node`.

```C++
auto heartbeat_pub = node_hdl.create_publisher<uavcan::node::Heartbeat>(/* to be defined */);
heartbeat_pub.publish(heartbeat_msg);
```

```C++
auto heartbeat_sub = node_hdl.::create_subscription<uavcan::node::Heartbeat>(/* to be defined */);
```

pavel-kirienko, 2022-12-13:
This is especially interesting for subscribers because it allows you to implement type erasure on the callback, thus allowing the customer to pass arbitrary functions/lambdas as handlers.

aentinger, 2022-12-13:
Lambdas are the plan :grin: . If there some specific API you are thinking of, feel free to propose it.

PR #182 (2023-01-09): Redesign library API around dedicated objects: `Publisher`, `Subscriber`, etc.
This fixes #174.
closed 2023-01-09 by aentinger