107-systems/107-Arduino-Cyphal #189: Fix UniqueId for ATSAME5x

generationmake, 2023-01-03:
[image]

### :bug: Bug Report

I think UniqueId does not work with ATSAME5x (https://www.adafruit.com/product/4759).
I think it goes into ARDUINO_ARCH_SAMD but there is nothing defined for ATSAME5x.

aentinger, 2023-01-03:
A very quick and dirty fix, try changing this [line](https://github.com/107-systems/107-Arduino-Cyphal/blob/main/src/utility/UniqueId16-samd.cpp#L14):

```diff
-#ifdef ARDUINO_ARCH_SAMD
+#if defined(ARDUINO_ARCH_SAMD) || defined(ARDUINO_SAMD_ADAFRUIT)
```

aentinger, 2023-01-03:
And also this [line](https://github.com/107-systems/107-Arduino-Cyphal/blob/main/src/utility/UniqueId16.cpp#L27) accordingly.

generationmake, 2023-01-03:
hi @aentinger 
it seems like it is more a problem with this line: https://github.com/107-systems/107-Arduino-Cyphal/blob/main/src/utility/UniqueId16.h#L38

aentinger, 2023-01-03:
Done @generationmake :heavy_check_mark: :wink:

commit 4939777a (2023-01-03): Fix: Adafruit SAMD Core does not use ArduinoCore-API. (#190)

This fixes #189.
closed 2023-01-03 by aentinger