107-systems/107-Arduino-Cyphal #276: [Feature Request]

bakul14, 2024-07-03:
[image]

### ⚡ Feature Request

It is necessary to implement setting the priority subject and service messages, since this feature is available in libcanard.
Here is a piece of implementation of the publish(T const & msg) method. Priotiry is not a parameter and it is set to CanardPriorityNominal.

(Replace This Text: A concise description of your preferred solution. Things to address include:
* Details of the technical implementation
* Tradeoffs made in design decisions
* Caveats and considerations for the future

If there are multiple solutions, please present each one separately. Save comparisons for the very end.)

aentinger, 2024-07-03:
I am willing to merge a high quality pull request providing this functionality using the first approach.

bakul14, 2024-07-03:
Okay, it's good. I am ready to check your PR

aentinger, 2024-07-03:
I am willing to receive your PR ;)

pavel-kirienko, 2024-07-03:
@bakul14 that's not how it works. You need to fork this repository, push your changes there, then send a pull request: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

[image]

closed 2024-09-16 by aentinger