107-systems/107-Arduino-Cyphal #279: [Feature Request] Add metadata to service respone

edu-rossrobotics, 2024-08-14:
[image]

### ⚡ Feature Request

**Please Describe The Problem To Be Solved**
I have a board that needs to send Execute Command request to different boards and handle the response based on who is answering.

**(Optional): Suggest A Solution**
I've worked on a solution that allows to use a response callback function that contains the metadata so you can use it to "filter" the response based on the node. I'm not sure if this is the best approach nor if it make sense based on the design philosophy of the library. 

If you think it can be useful, I already have it working so I'm happy to open a PR to add it.

Thanks.

aentinger, 2024-08-14:
I believe the primary design of services is to facilitate request-response style bidirectional communication between primarily two communication partners. On the other hand, `GetInfo` is definitely designed for multiple recipients (AFAIK). @pavel-kirienko what's your take?

pavel-kirienko, 2024-08-16:
>I believe the primary design of services is to facilitate request-response style bidirectional communication between primarily two communication partners.

Correct.

>On the other hand, GetInfo is definitely designed for multiple recipients (AFAIK).

Nope, it's also request-response. You can snoop on third party responses if you want but it's not, strictly speaking, how it is supposed to be used.

edu-rossrobotics, 2024-08-16:
So, in that case how would you handle a situation like this?

You have a device A sending the same service request to 2 different devices (B and C). When you receive the responses back, one has succeed while the other one failed. How can you know which device sent each response? Are you assuming the responses will arrive in the same order you sent the request?

pavel-kirienko, 2024-08-16:
The library should provide the user with sufficient context to distinguish the two responses.

aentinger, 2024-08-17:
Well, I suppose let's have a look at your code then. How about preparing a PR showing your changes?

PR #280 (2024-08-18): Feature/service response with metadata
This PR closes #279. It uses the same structure used in subscriptions to allow the user to use the metadata info in the received response callback.
closed 2024-08-18 by aentinger